openbmc/dbus-sensors #7: Exported dbus sensor interfaces do not comply with dbus spec

jk-ozlabs, 2020-06-17:
The phosphor-dbus-interfaces spec defines sensor values as `int64`. From [the spec](https://github.com/openbmc/phosphor-dbus-interfaces/blob/12a8ca89d996ecebe09f70d107d8a63f9cbd441a/xyz/openbmc_project/Sensor/Value.interface.yaml#L20) :

```properties:
    - name: Value
      type: int64
      description: >
          The sensor reading.
```

However, the dbus-sensors implementation exports values as `double`:

```# busctl introspect xyz.openbmc_project.HwmonTempSensor /xyz/openbmc_project/sensors/temperature/Temp xyz.openbmc_project.Sensor.Value
NAME                             TYPE      SIGNATURE RESULT/VALUE FLAGS
.MaxValue                        property  d         127          emits-change
.MinValue                        property  d         -128         emits-change
.Value                           property  d         0            emits-change writable
```

This causes issues for processes trying to read the sensor values using the standard interface.

feistjj, 2020-06-17:
Fix is here https://gerrit.openbmc-project.xyz/c/openbmc/phosphor-dbus-interfaces/+/11739

jk-ozlabs, 2020-06-18:
Brilliant, thanks. +1 from me.

jk-ozlabs, 2020-07-07:
Fixed in https://github.com/openbmc/phosphor-dbus-interfaces/commit/4ec6587fa908fe9df41cc0962ab6ff6446fbd54f , closing.

closed 2020-07-07 by jk-ozlabs