openbmc/dbus-sensors #13: README.md hyperlink ExternalSensor returns a 404 ERROR

BMC-Bruce, 2021-05-07:
[ExternalSensor](https://github.com/openbmc/docs/blob/master/designs/external-sensor.mdl) virtual sensor
returns a 404 ERROR.

Also note there already exists a [virtual-sensors.md](https://github.com/openbmc/docs/blob/master/designs/virtual-sensors.md)
in the [docs/design](https://github.com/openbmc/docs/tree/master/designs) folder.  Maybe _virtual sensor_, after ExternalSensor,
should have different wording.

edtanous, 2021-07-20:
did this link get fixed?

BMC-Bruce, 2021-08-31:
No, just check I still get 404 ERROR.

BMC-Bruce, 2021-09-01:
I feel like this needs to be assigned to Josh Lehan Krellan, based off of when I found the issue.

williamspatrick, 2021-09-01:
The issue is that the design was never merged in docs.  https://gerrit.openbmc-project.xyz/c/openbmc/docs/+/41452  I just merged it so this can probably be closed.

BMC-Bruce, 2021-09-01:
Thank you Patrick (and Josh for writing it), that does resolve this issue!

closed 2021-09-01 by BMC-Bruce