openbmc/dbus-sensors #14: ExitAirTempSensor not settable

zkurzyns, 2021-07-09:
After this change https://github.com/openbmc/dbus-sensors/commit/1263c3dad3a723f4ebbd05e4031e4d5ff45f349c some of my tests using `busctl set-property` are not able to change value of the /xyz/openbmc_project/sensors/cfm/System_Airflow sensor exposed by xyz.openbmc_project.ExitAirTempSensor service.
(In other words, this sensor doesn't react on SpecialMode==Manufacturing)

How I can set the value of this sensor after this change?
If this sensor was intentionally omitted then what is the reason behind it?

edtanous, 2021-07-20:
I don't believe that sensor was explicitly omitted.  @Bruceleequantatw can you take a look at this?

Bruceleequantatw, 2021-07-21:
ExitAirTempSensor is a not settable sensor, but when _manufacturingModeStatus == "xyz.openbmc_project.Control.Security.SpecialMode.Modes.Manufacturing"_ (https://github.com/openbmc/dbus-sensors/blob/master/src/Utils.cpp#L537) the _"manufacturingMode" will be "true"_ and then allow to set sensor value
(https://github.com/openbmc/dbus-sensors/blob/master/include/sensor.hpp#L183), please check the _SpecialMode == "xyz.openbmc_project.Control.Security.SpecialMode.Modes.Manufacturing"_.

zkurzyns, 2021-07-21:
The method you mentioned here: (https://github.com/openbmc/dbus-sensors/blob/master/src/Utils.cpp#L537) named: **setupManufacturingModeMatch** is called from the main method of most sensors in this code, but not in case of the ExitAirTempSensor. 
This makes the sensor unresponsive to changes of the SpecialMode and it cannot be set manually.
See its main method: https://github.com/openbmc/dbus-sensors/blob/3453354b838885bfe47754bdf116ecedde87116d/src/ExitAirTempSensor.cpp#L963

Bruceleequantatw, 2021-07-22:
This will be fixed after code review, thanks.
https://gerrit.openbmc-project.xyz/c/openbmc/dbus-sensors/+/45136

zkurzyns, 2021-07-22:
Issue solved, question answered, closing.
Thank you for support.

closed 2021-07-22 by zkurzyns