openbmc/dbus-sensors #18: NVMe sensors reading issue

operezmuena, 2022-03-30:
I've recently began testing NVMe sensors and I've been having temperature reading issues with my Intel SSD P4510 devices.  Debugging the NVMe code, I noticed that the "i2c_smbus_read_block_data()" function is returning a size of 6 but, no sensor data is found in the "resp" vector. 

`size = i2c_smbus_read_block_data(fileHandle.handle(), cmd, resp.data());`

Only after I changed 'resp' from being a std::vector type to a std::array type, I began seeing valid temperature readings from my SSD drives.  Here is my patch for you reference:

```
From c72120055b73f0a328464f971c60e6ad0760e449 Mon Sep 17 00:00:00 2001
From: Oscar A Perez <oscar.perez@ztsystems.com>
Date: Wed, 30 Mar 2022 19:40:23 +0000
Subject: [PATCH] NVMe temperature readings not working

---
 src/NVMeBasicContext.cpp | 29 +++++++++++++++++++----------
 1 file changed, 19 insertions(+), 10 deletions(-)

diff --git a/src/NVMeBasicContext.cpp b/src/NVMeBasicContext.cpp
index 08333c59..ef264f8e 100644
--- a/src/NVMeBasicContext.cpp
+++ b/src/NVMeBasicContext.cpp
@@ -16,6 +16,7 @@
 #include <cstring>
 #include <system_error>
 #include <thread>
+#include <span>

 extern "C"
 {
@@ -59,8 +60,9 @@ static void decodeBasicQuery(const std::array<uint8_t, 6>& req, int& bus,
     offset = req[sizeof(busle) + 1];
 }

+template<std::size_t SIZE>
 static ssize_t execBasicQuery(int bus, uint8_t addr, uint8_t cmd,
-                              std::vector<uint8_t>& resp)
+                              std::array<uint8_t, SIZE>& resp)
 {
     int32_t size = 0;
     std::filesystem::path devpath = "/dev/i2c-" + std::to_string(bus);
@@ -77,8 +79,6 @@ static ssize_t execBasicQuery(int bus, uint8_t addr, uint8_t cmd,
         return -errno;
     }

-    resp.reserve(UINT8_MAX + 1);
-
     /* Issue the NVMe MI basic command */
     size = i2c_smbus_read_block_data(fileHandle.handle(), cmd, resp.data());
     if (size < 0)
@@ -101,7 +101,7 @@ static ssize_t execBasicQuery(int bus, uint8_t addr, uint8_t cmd,

 static ssize_t processBasicQueryStream(FileHandle& in, FileHandle& out)
 {
-    std::vector<uint8_t> resp{};
+    std::array<uint8_t, UINT8_MAX + 1> resp{};
     ssize_t rc = 0;

     while (true)
@@ -164,11 +164,11 @@ static ssize_t processBasicQueryStream(FileHandle& in, FileHandle& out)
         }

         /* Write out the response data */
-        std::vector<uint8_t>::iterator cursor = resp.begin();
-        while (cursor != resp.end())
+        std::span<uint8_t> mySpan{resp.data(), len};
+        for (auto it = mySpan.begin(); it != mySpan.end(); )
         {
-            size_t lenRemaining = std::distance(cursor, resp.end());
-            ssize_t egress = ::write(out.handle(), &(*cursor), lenRemaining);
+            size_t lenRemaining = std::distance(it, mySpan.end());
+            ssize_t egress = ::write(out.handle(), &(*it), lenRemaining);
             if (egress == -1)
             {
                 std::cerr << "Failed to write block data of length " << std::dec
@@ -180,8 +180,7 @@ static ssize_t processBasicQueryStream(FileHandle& in, FileHandle& out)
                 }
                 return -EIO;
             }
-
-            cursor += egress;
+            it += egress;
         }
     }

@@ -398,6 +397,16 @@ void NVMeBasicContext::processResponse(std::shared_ptr<NVMeSensor>& sensor,

     uint8_t* messageData = static_cast<uint8_t*>(msg);

+#if DEBUG
+    std::cerr << "\nmessageData: ";
+    for (size_t i=0; i < len; i++)
+    {
+        std::cerr << std::hex << std::setfill('0') << std::setw(2)
+                  << static_cast<int>(messageData[i]) << " ";
+    }
+    std::cerr << "\n";
+#endif
+
     uint8_t status = messageData[0];
     if ((status & NVME_MI_BASIC_SFLGS_DRIVE_NOT_READY) ||
         !(status & NVME_MI_BASIC_SFLGS_DRIVE_FUNCTIONAL))
--
2.17.1

```

operezmuena, 2022-03-31:
@amboar  I just tested your patch and it does solve my issues.  Thank you for the prompt response.

amboar, 2022-03-31:
Actually, this was broken by 73030639a5ba8 at https://github.com/openbmc/dbus-sensors/commit/73030639a5ba8#diff-5abbed76e66786eec2a94b186f608b090c253722a2b37635b278e0740e3a5e4aL185-L187 where we switch from accessing the backing buffer directly to using an iterator over the vector.

commit a7afacc4 (2022-04-01): NVMeBasicContext: Fix response vector size

73030639a5ba ("Enable cppcoreguidelines-avoid-goto checks") switches
from passing the backing store of the response vector to ::write() to
using an iterator over the response vector. This exposed the flaw that
the vector was never properly sized to accommodate the presence of the
response in the backing buffer[1], though from a memory-safety
perspective the backing buffer _was_ appropriately sized from the
reserve() call.

Resize the vector both before and after to ensure that a complete
response can be captured safely while also returning a vector t [... cut]
closed 2022-04-01 by bradbishop