openbmc/dbus-sensors #20: ASAN reports use-after-free in hwmontempsensor

zevweiss, 2022-07-22:
I'm working on some patches to allow dbus-sensors daemons to manage driver binding/unbinding on host power state transitions, which involves an increase in the amount of sensor object creation & destruction at runtime (post initialization), and started seeing some ASAN reports that didn't seem to be coming from my local changes; after some investigation I _think_ there's a latent bug in the existing upstream code that just requires a bit more activity to trigger.

I reduced my local patch to the following, just enabling ASAN and adding a dbus method to trigger a `createSensors()` call:

```diff
diff --git a/meson.build b/meson.build
index 8b29e64aa4a5..dd675b8c98de 100644
--- a/meson.build
+++ b/meson.build
@@ -24,6 +24,7 @@ uring_args = [
 add_project_arguments(
     '-Wno-psabi',
     '-Wuninitialized',
+    '-fsanitize=address',
     '-DBOOST_SYSTEM_NO_DEPRECATED',
     '-DBOOST_ERROR_CODE_HEADER_ONLY',
     '-DBOOST_NO_RTTI',
diff --git a/src/HwmonTempMain.cpp b/src/HwmonTempMain.cpp
index d1d50d6a47c9..36b96b0fc05a 100644
--- a/src/HwmonTempMain.cpp
+++ b/src/HwmonTempMain.cpp
@@ -583,5 +583,15 @@ int main()
 
     matches.emplace_back(std::move(ifaceRemovedMatch));
 
+    std::shared_ptr<sdbusplus::asio::dbus_interface> debugIface =
+      objectServer.add_interface("/xyz/openbmc_project/sensors/temperature",
+                                 "xyz.openbmc_project.Debug");
+    debugIface->register_method("Rebuild", [&]() {
+      std::cerr << "Rebuilding sensors...\n";
+      createSensors(io, objectServer, sensors, systemBus, nullptr);
+      std::cerr << "...done.\n";
+    });
+    debugIface->initialize();
+
     io.run();
 }
diff --git a/src/meson.build b/src/meson.build
index 4c3d61cd6886..e340bb084655 100644
--- a/src/meson.build
+++ b/src/meson.build
@@ -85,6 +85,7 @@ if get_option('hwmon-temp').enabled()
             thresholds_dep,
             utils_dep,
         ],
+        link_args: '-fsanitize=address',
         cpp_args: uring_args,
         implicit_include_directories: false,
         include_directories: '../include',
```

With another shell open manually running a sequence of `busctl call xyz.openbmc_project.HwmonTempSensor /xyz/openbmc_project/sensors/temperature xyz.openbmc_project.Debug Rebuild` calls, I got the following ASAN dump:

```
root@romed8hm3-d05099f41e18:~# LD_LIBRARY_PATH=/tmp /tmp/hwmontempsensor
error getting  SpecialMode status No route to host
Rebuilding sensors...
...done.
Rebuilding sensors...
...done.
Rebuilding sensors...
...done.
Rebuilding sensors...
...done.
Rebuilding sensors...
...done.
=================================================================
==30937==ERROR: AddressSanitizer: heap-use-after-free on address 0x73d46080 at pc 0x006107f0 bp 0x7ef5e34c sp 0x7ef5e338
READ of size 1 at 0x73d46080 thread T0
    #0 0x6107ec in bool __gnu_cxx::__ops::_Iter_equals_val<char const>::operator()<boost::asio::buffers_iterator<boost::asio::const_buffers_1, char> >(boost::asio::buffers_iterator<boost::asio::const_buffers_1, char>) /usr/include/c++/11.3.0/bits/predefined_ops.h:270
[... 121 lines cut ...]
  0x2e7a8c60: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==30937==ABORTING
```

(The `LD_LIBRARY_PATH` setting is just to allow it to find `libasan.so.6`, which I manually copied in after building the `gcc-sanitizers` package.)

zevweiss, 2022-07-22:
On a suggestion from @edtanous I tried applying the following patch so that the `readBuf` destructor is ordered after the `inputDev` destructor:

```diff
diff --git a/include/HwmonTempSensor.hpp b/include/HwmonTempSensor.hpp
index 5cf77c384db0..8e478fca280a 100644
--- a/include/HwmonTempSensor.hpp
+++ b/include/HwmonTempSensor.hpp
@@ -36,9 +36,9 @@ class HwmonTempSensor :
 
   private:
     sdbusplus::asio::object_server& objServer;
+    boost::asio::streambuf readBuf;
     boost::asio::posix::stream_descriptor inputDev;
     boost::asio::deadline_timer waitTimer;
-    boost::asio::streambuf readBuf;
     std::string path;
     double offsetValue;
     double scaleValue;
```

I was still able to reproduce the problem, however (I believe this is functionally pretty much identical to the dump above, but I'll preserve it here in case there's a difference I've missed):

```
root@romed8hm3-d05099f41e18:~# LD_LIBRARY_PATH=/tmp /tmp/hwmontempsensor
error getting  SpecialMode status No route to host
Rebuilding sensors...
...done.
Rebuilding sensors...
...done.
Rebuilding sensors...
...done.
Rebuilding sensors...
...done.
Rebuilding sensors...
...done.
Rebuilding sensors...
...done.
Rebuilding sensors...
...done.
Rebuilding sensors...
...done.
Rebuilding sensors...
...done.
Rebuilding sensors...
...done.
Rebuilding sensors...
...done.
Rebuilding sensors...
...done.
=================================================================
==31018==ERROR: AddressSanitizer: heap-use-after-free on address 0x73c85280 at pc 0x005e06d8 bp 0x7ee1034c sp 0x7ee10338
READ of size 1 at 0x73c85280 thread T0
    #0 0x5e06d4 in bool __gnu_cxx::__ops::_Iter_equals_val<char const>::operator()<boost::asio::buffers_iterator<boost::asio::const_buffers_1, char> >(boost::asio::buffers_iterator<boost::asio::const_buffers_1, char>) /usr/include/c++/11.3.0/bits/predefined_ops.h:270
    #1 0x5e06d4 in boost::asio::buffers_iterator<boost::asio::const_buffers_1, char> std::__find_if<boost::asio::buffers_iterator<boost::asio::const_buffers_1, char>, __gnu_cxx::__ops::_Iter_equals_val<char const> >(boost::asio::buffers_iterator<boost::asio::const_buffers_1, char>, boost::asio::buffers_iterator<boost::asio::const_buffers_1, char>, __gnu_cxx::__ops::_Iter_equals_val<char const>, std::random_access_iterator_tag) /usr/include/c++/11.3.0/bits/stl_algobase.h:2069
    #2 0x5e06d4 in boost::asio::buffers_iterator<boost::asio::const_buffers_1, char> std::__find_if<boost::asio::buffers_iterator<boost::asio::const_buffers_1, char>, __gnu_cxx::__ops::_Iter_equals_val<char const> >(boost::asio::buffers_iterator<boost::asio::const_buffers_1, char>, boost::asio::buffers_iterator<boost::asio::const_buffers_1, char>, __gnu_cxx::__ops::_Iter_equals_val<char const>) /usr/include/c++/11.3.0/bits/stl_algobase.h:2114
    #3 0x5e06d4 in boost::asio::buffers_iterator<boost::asio::const_buffers_1, char> std::find<boost::asio::buffers_iterator<boost::asio::const_buffers_1, char>, char>(boost::asio::buffers_iterator<boost::asio::const_buffers_1, char>, boost::asio::buffers_iterator<boost::asio::const_buffers_1, char>, char const&) /usr/include/c++/11.3.0/bits/stl_algo.h:3884
    #4 0x5e06d4 in operator() /usr/include/boost/asio/impl/read_until.hpp:874
    #5 0x5e582c in operator() /usr/include/boost/asio/detail/bind_handler.hpp:289
    #6 0x5e582c in asio_handler_invoke<boost::asio::detail::binder2<boost::asio::detail::read_until_delim_op_v1<boost::asio::posix::basic_stream_descriptor<>, boost::asio::basic_streambuf_ref<std::allocator<char> >, HwmonTempSensor::setupRead()::<lambda(const boost::system::error_code&, std::size_t)> >, boost::system::error_code, unsigned int> > /usr/include/boost/asio/handler_invoke_hook.hpp:88
    #7 0x5e582c in invoke<boost::asio::detail::binder2<boost::asio::detail::read_until_delim_op_v1<boost::asio::posix::basic_stream_descriptor<>, boost::asio::basic_streambuf_ref<std::allocator<char> >, HwmonTempSensor::setupRead()::<lambda(const boost::system::error_code&, std::size_t)> >, boost::system::error_code, unsigned int>, HwmonTempSensor::setupRead()::<lambda(const boost::system::error_code&, std::size_t)> > /usr/include/boost/asio/detail/handler_invoke_helpers.hpp:54
    #8 0x5e582c in asio_handler_invoke<boost::asio::detail::binder2<boost::asio::detail::read_until_delim_op_v1<boost::asio::posix::basic_stream_descriptor<>, boost::asio::basic_streambuf_ref<std::allocator<char> >, HwmonTempSensor::setupRead()::<lambda(const boost::system::error_code&, std::size_t)> >, boost::system::error_code, unsigned int>, boost::asio::posix::basic_stream_descriptor<>, boost::asio::basic_streambuf_ref<std::allocator<char> >, HwmonTempSensor::setupRead()::<lambda(const boost::system::error_code&, std::size_t)> > /usr/include/boost/asio/impl/read_until.hpp:993
    #9 0x5e582c in invoke<boost::asio::detail::binder2<boost::asio::detail::read_until_delim_op_v1<boost::asio::posix::basic_stream_descriptor<>, boost::asio::basic_streambuf_ref<std::allocator<char> >, HwmonTempSensor::setupRead()::<lambda(const boost::system::error_code&, std::size_t)> >, boost::system::error_code, unsigned int>, boost::asio::detail::read_until_delim_op_v1<boost::asio::posix::basic_stream_descriptor<>, boost::asio::basic_streambuf_ref<std::allocator<char> >, HwmonTempSensor::setupRead()::<lambda(const boost::system::error_code&, std::size_t)> > > /usr/include/boost/asio/detail/handler_invoke_helpers.hpp:54
[... cut ...]
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==31018==ABORTING
```

zevweiss, 2022-07-25:
I reproduced it again with the following patch to add some tracking of what's going on:

```diff
diff --git a/meson.build b/meson.build
index dd675b8c98de..19001f01f907 100644
--- a/meson.build
+++ b/meson.build
@@ -32,6 +32,7 @@ add_project_arguments(
     '-DBOOST_ALL_NO_LIB',
     '-DBOOST_ASIO_DISABLE_THREADS',
     '-DBOOST_ALLOW_DEPRECATED_HEADERS',
+    '-DBOOST_ASIO_ENABLE_HANDLER_TRACKING',
     language: 'cpp',
 )
 
diff --git a/src/HwmonTempSensor.cpp b/src/HwmonTempSensor.cpp
index 7eb5e8fc1196..f34fe01efa1c 100644
--- a/src/HwmonTempSensor.cpp
+++ b/src/HwmonTempSensor.cpp
@@ -88,6 +88,7 @@ HwmonTempSensor::HwmonTempSensor(
 
 HwmonTempSensor::~HwmonTempSensor()
 {
+    fprintf(stderr, "~HwmonTempSensor(%p, inputDev=%p)\n", (void*)this, (void*)&inputDev);
     // close the input dev to cancel async operations
     inputDev.close();
     waitTimer.cancel();
@@ -97,6 +98,7 @@ HwmonTempSensor::~HwmonTempSensor()
     }
     objServer.remove_interface(sensorInterface);
     objServer.remove_interface(association);
+    fprintf(stderr, "~HwmonTempSensor(%p) done\n", (void*)this);
 }
 
 void HwmonTempSensor::setupRead(void)
```

The full resulting dump is attached [here](https://github.com/openbmc/dbus-sensors/files/9184627/asio-debug-2.txt).  Annotating a relevant-looking portion toward the end:

```
...
    420 @asio|1658780727.500022|>99|ec=system:0
    421 @asio|1658780727.500369|99^117|in 'async_read_until' (/usr/src/debug/dbus-sensors/0.1+git999-r0/recipe-sysroot/usr/include/boost/asio/:909)
    422 @asio|1658780727.500369|99*117|descriptor@0x73c0ea5c.async_read_some # handler 117 created
    423 @asio|1658780727.500793|<99|
    424 @asio|1658780727.511663|>113|
    425 ~HwmonTempSensor(0x73c0f810, inputDev=0x73c0f958)
    426 @asio|1658780745.566319|113|descriptor@0x73c0f95c.close
    427 @asio|1658780745.568109|113|deadline_timer@0x73c0f978.cancel
    428 ~HwmonTempSensor(0x73c0f810) done
    429 @asio|1658780746.800263|113^118|in 'async_read_until' (/usr/src/debug/dbus-sensors/0.1+git999-r0/recipe-sysroot/usr/include/boost/asio/:909)
    430 @asio|1658780746.800263|113*118|descriptor@0x73c1a35c.async_read_some
    431 ~HwmonTempSensor(0x73c0e910, inputDev=0x73c0ea58)
    432 @asio|1658780746.801711|113|descriptor@0x73c0ea5c.close # descriptor for handler 117 closed
    433 @asio|1658780746.815146|113|deadline_timer@0x73c0ea78.cancel
    434 ~HwmonTempSensor(0x73c0e910) done
    435 @asio|1658780747.881132|113^119|in 'async_read_until' (/usr/src/debug/dbus-sensors/0.1+git999-r0/recipe-sysroot/usr/include/boost/asio/:909)
    436 @asio|1658780747.881132|113*119|descriptor@0x73c1945c.async_read_some
    437 ~HwmonTempSensor(0x73c0dc90, inputDev=0x73c0ddd8)
    438 @asio|1658780748.012149|113|descriptor@0x73c0dddc.close
    439 @asio|1658780748.012529|113|deadline_timer@0x73c0ddf8.cancel
    440 ~HwmonTempSensor(0x73c0dc90) done
    441 @asio|1658780749.360577|113^120|in 'async_read_until' (/usr/src/debug/dbus-sensors/0.1+git999-r0/recipe-sysroot/usr/include/boost/asio/:909)
    442 @asio|1658780749.360577|113*120|descriptor@0x73c187dc.async_read_some
    443 ~HwmonTempSensor(0x73c0cb10, inputDev=0x73c0cc58)
    444 @asio|1658780749.362053|113|descriptor@0x73c0cc5c.close
    445 @asio|1658780749.362425|113|deadline_timer@0x73c0cc78.cancel
    446 ~HwmonTempSensor(0x73c0cb10) done
    447 @asio|1658780750.406205|113^121|in 'async_read_until' (/usr/src/debug/dbus-sensors/0.1+git999-r0/recipe-sysroot/usr/include/boost/asio/:909)
    448 @asio|1658780750.406205|113*121|descriptor@0x73c178dc.async_read_some
    449 ~HwmonTempSensor(0x73c0c110, inputDev=0x73c0c258)
    450 @asio|1658780750.407679|113|descriptor@0x73c0c25c.close
    451 @asio|1658780750.408051|113|deadline_timer@0x73c0c278.cancel
    452 ~HwmonTempSensor(0x73c0c110) done
    453 @asio|1658780751.444188|113^122|in 'async_read_until' (/usr/src/debug/dbus-sensors/0.1+git999-r0/recipe-sysroot/usr/include/boost/asio/:909)
    454 @asio|1658780751.444188|113*122|descriptor@0x73c1ebdc.async_read_some
    455 ~HwmonTempSensor(0x73c13190, inputDev=0x73c132d8)
    456 @asio|1658780751.561851|113|descriptor@0x73c132dc.close
    457 @asio|1658780751.562224|113|deadline_timer@0x73c132f8.cancel
    458 ~HwmonTempSensor(0x73c13190) done
    459 @asio|1658780752.380922|113^123|in 'async_read_until' (/usr/src/debug/dbus-sensors/0.1+git999-r0/recipe-sysroot/usr/include/boost/asio/:909)
    460 @asio|1658780752.380922|113*123|descriptor@0x73c1df5c.async_read_some
    461 ~HwmonTempSensor(0x73c11b10, inputDev=0x73c11c58)
    462 @asio|1658780752.387451|113|descriptor@0x73c11c5c.close
    463 @asio|1658780752.387824|113|deadline_timer@0x73c11c78.cancel
    464 ~HwmonTempSensor(0x73c11b10) done
    465 @asio|1658780752.999959|113^124|in 'async_read_until' (/usr/src/debug/dbus-sensors/0.1+git999-r0/recipe-sysroot/usr/include/boost/asio/:909)
    466 @asio|1658780752.999959|113*124|descriptor@0x73c1cb5c.async_read_some
    467 @asio|1658780753.113799|113*125|io_context@0x7eb068b0.execute
    468 @asio|1658780753.114080|<113|
    469 @asio|1658780753.114522|>117|ec=system:0,bytes_transferred=6 # handler 117 entered
    470 =================================================================
    471 ==1593==ERROR: AddressSanitizer: heap-use-after-free on address 0x73c0df00 at pc 0x005c2294 bp 0x7eb0514c sp 0x7eb05138
...
```

So:
 1. At line 422 we see handler 117 created for descriptor 0x73c0ea5c
 2. At line 432 we see descriptor 0x73c0ea5c closed
 3. At line 469 handler 117 is invoked

It looks like [the documented behavior of `boost::asio::posix::stream_descriptor::close()`](https://www.boost.org/doc/libs/1_79_0/doc/html/boost_asio/reference/posix__basic_stream_descriptor/close/overload1.html) isn't actually happening, as far as I can see?  (No `~` actions or handlers receiving `boost::asio::error::operation_aborted` in the asio trace.)

zevweiss, 2022-07-29:
As an experiment I tried running with the following patch to re-do hwmontempsensor's read buffering as in [this patch by @edtanous](https://gerrit.openbmc.org/c/openbmc/dbus-sensors/+/55041) and have thus far been unable to reproduce the problem.

```diff
diff --git a/include/HwmonTempSensor.hpp b/include/HwmonTempSensor.hpp
index 8e478fca280a..ee26b24a5f08 100644
--- a/include/HwmonTempSensor.hpp
+++ b/include/HwmonTempSensor.hpp
@@ -1,7 +1,7 @@
 #pragma once
 
 #include <Thresholds.hpp>
-#include <boost/asio/streambuf.hpp>
+#include <boost/asio/random_access_file.hpp>
 #include <sdbusplus/asio/object_server.hpp>
 #include <sensor.hpp>
 
@@ -36,15 +36,15 @@ class HwmonTempSensor :
 
   private:
     sdbusplus::asio::object_server& objServer;
-    boost::asio::streambuf readBuf;
-    boost::asio::posix::stream_descriptor inputDev;
+    std::array<char, 128> readBuf;
+    boost::asio::random_access_file inputDev;
     boost::asio::deadline_timer waitTimer;
     std::string path;
     double offsetValue;
     double scaleValue;
     unsigned int sensorPollMs;
 
-    void handleResponse(const boost::system::error_code& err);
+    void handleResponse(const boost::system::error_code& err, size_t bytesTransferred);
     void restartRead();
     void checkThresholds(void) override;
 };
diff --git a/src/HwmonTempSensor.cpp b/src/HwmonTempSensor.cpp
index f34fe01efa1c..b478303a1588 100644
--- a/src/HwmonTempSensor.cpp
+++ b/src/HwmonTempSensor.cpp
@@ -23,6 +23,7 @@
 #include <sdbusplus/asio/connection.hpp>
 #include <sdbusplus/asio/object_server.hpp>
 
+#include <charconv>
 #include <iostream>
 #include <istream>
 #include <limits>
@@ -51,20 +52,12 @@ HwmonTempSensor::HwmonTempSensor(
            std::move(thresholdsIn), sensorConfiguration, objectType, false,
            false, thisSensorParameters.maxValue, thisSensorParameters.minValue,
            conn, powerState),
-    objServer(objectServer), inputDev(io), waitTimer(io), path(path),
+    objServer(objectServer), inputDev(io, path, boost::asio::random_access_file::read_only),
+    waitTimer(io), path(path),
     offsetValue(thisSensorParameters.offsetValue),
     scaleValue(thisSensorParameters.scaleValue),
     sensorPollMs(static_cast<unsigned int>(pollRate * 1000))
 {
-    // NOLINTNEXTLINE(cppcoreguidelines-pro-type-vararg)
-    int fd = open(path.c_str(), O_RDONLY);
-    if (fd < 0)
-    {
-        std::cerr << "HwmonTempSensor " << sensorName << " failed to open "
-                  << path << "\n";
-    }
-    inputDev.assign(fd);
-
     sensorInterface = objectServer.add_interface(
         "/xyz/openbmc_project/sensors/" + thisSensorParameters.typeName + "/" +
             name,
@@ -112,13 +105,13 @@ void HwmonTempSensor::setupRead(void)
     }
 
     std::weak_ptr<HwmonTempSensor> weakRef = weak_from_this();
-    boost::asio::async_read_until(inputDev, readBuf, '\n',
-                                  [weakRef](const boost::system::error_code& ec,
-                                            std::size_t /*bytes_transfered*/) {
+    inputDev.async_read_some_at(0, boost::asio::buffer(readBuf),
+                                [weakRef](const boost::system::error_code& ec,
+                                          std::size_t bytesTransferred) {
         std::shared_ptr<HwmonTempSensor> self = weakRef.lock();
         if (self)
         {
-            self->handleResponse(ec);
+            self->handleResponse(ec, bytesTransferred);
         }
     });
 }
@@ -141,7 +134,7 @@ void HwmonTempSensor::restartRead()
     });
 }
 
-void HwmonTempSensor::handleResponse(const boost::system::error_code& err)
+void HwmonTempSensor::handleResponse(const boost::system::error_code& err, size_t bytesTransferred)
 {
     if ((err == boost::system::errc::bad_file_descriptor) ||
         (err == boost::asio::error::misc_errors::not_found))
@@ -150,20 +143,19 @@ void HwmonTempSensor::handleResponse(const boost::system::error_code& err)
                   << "\n";
         return; // we're being destroyed
     }
-    std::istream responseStream(&readBuf);
+
     if (!err)
     {
-        std::string response;
-        std::getline(responseStream, response);
-        try
+        const char* bufEnd = readBuf.data() + bytesTransferred;
+        int nvalue = 0;
+        std::from_chars_result ret = std::from_chars(readBuf.data(), bufEnd, nvalue);
+        if (ret.ec != std::errc())
         {
-            rawValue = std::stod(response);
-            double nvalue = (rawValue + offsetValue) * scaleValue;
-            updateValue(nvalue);
+            incrementError();
         }
-        catch (const std::invalid_argument&)
+        else
         {
-            incrementError();
+            updateValue((nvalue + offsetValue) * scaleValue);
         }
     }
     else
@@ -171,7 +163,6 @@ void HwmonTempSensor::handleResponse(const boost::system::error_code& err)
         incrementError();
     }
 
-    responseStream.clear();
     inputDev.close();
 
     // NOLINTNEXTLINE(cppcoreguidelines-pro-type-vararg)
```

I would certainly like to understand what's going wrong with the existing buffering scheme (ASIO bug?  some subtle misuse of its API?), but this might be a viable alternative that was probably likely to get done at some point in the not-too-distant future anyway.

edtanous, 2022-07-29:
> I would certainly like to understand what's going wrong with the existing buffering scheme 

IMO if we have a fix, and it has no downsides, we should just fix it on master, then chasing down the old bug becomes an academic excersize, which, we might want to do for learning on how to not find that issue again, but in the meantime, the core issue is fixed.

Get the above patch submitted to gerrit, and lets see if we can get it cleaned up and on master.

zevweiss, 2022-07-29:
Patch posted here: https://gerrit.openbmc.org/c/openbmc/dbus-sensors/+/56019

zevweiss, 2022-08-04:
While the root cause remains mysterious, commit [d1c8f44](https://github.com/openbmc/dbus-sensors/commit/d1c8f4433e981696d06e3c29006affd96624c89e) appears to resolve this.

closed 2022-08-04 by zevweiss