openbmc/dbus-sensors #23: NVMeSensor Missing ObjectManager

iwoloschin, 2023-01-13:
The NVMeSensor daemon is missing an ObjectManager at `/xyz/openbmc_project/sensors`, it is instead incorrectly at `/xyz/openbmc_sensors/`. This results in no NVMe temperature sensors being available.

https://github.com/openbmc/dbus-sensors/blob/master/src/NVMeSensorMain.cpp#L235

I've tested removing the trailing `/` locally and it works on Micron 7450 NVMe drives I happened to have in my system. I do not have a CCLA signed yet so I can't technically submit a patch so this is the best I can do for the moment.

amboar, 2023-01-17:
@iwoloschin please test https://gerrit.openbmc.org/c/openbmc/dbus-sensors/+/60299 and mark it as verified if it works for you.

iwoloschin, 2023-01-17:
Thanks @amboar. I made the same patch locally last week before filing this bug and it resolved the issue.

commit ea148ec6 (2023-01-17): NVMeSensorMain: Fix sensor object manager path

Change 661d4378467a ("Identify active sensors via the D-Bus "sensors"
service") added a trailing '/' to the path. Remove it, as it breaks
sensor detection by introspection.

Resolves: #23
Fixes: 661d4378467a ("Identify active sensors via the D-Bus "sensors" service")

Change-Id: I6fed1a36b6feae84160cdb5e16a1bdb7cebed262
closed 2023-01-17 by bradbishop