openbmc/dbus-sensors #24: hwmontempsensor crashes when given (mis) configuration for NCT7802 Sensor

pointbazaar, 2023-05-03:
```json
{
      "Address": "0x28",
      "Bus": 5,
      "Name": "GARBO_SENSOR",
      "Name1": "SYS_Air_Inlet",
      "Name2": "MB_Air_Inlet",
      "Name3": "MB_Air_Outlet",
      "Thresholds": [
        {
          "Direction": "greater than",
          "Label": "temp1",
          "Name": "upper critical",
          "Severity": 1,
          "Value": 40
        },
        {
          "Direction": "greater than",
          "Label": "temp2",
          "Name": "upper critical",
          "Severity": 1,
          "Value": 50
        },
        {
          "Direction": "greater than",
          "Label": "temp3",
          "Name": "upper critical",
          "Severity": 1,
          "Value": 70
        }
      ],
      "Type": "NCT7802"
    },
```

one solution is to remove the code for reading the thresholds from hwmontempsensor.
Then ofc no thresholds but atleast the sensors show up. 

the reason for crash is dbus related, it tries to create an interface which already exists. 
the interface in question is related to the thresholds.

pointbazaar, 2023-05-10:
When providing an empty Thresholds Array, it also works.

pointbazaar, 2023-05-10:
hwmontempsensor works when using Index based configuration instead of "Label" like in the above example.

Still not good that configuration can crash hwmontempsensor so i leave this open.

```
    {
      "Address": "0x28",
      "Bus": 5,
      "Name": "GARBO_SENSOR",
      "Name1": "SYS_Air_Inlet",
      "Name2": "MB_Air_Inlet",
      "Name3": "MB_Air_Outlet",
      "Thresholds": [
         {
           "Direction": "less than",
           "Name": "lower critical",
           "Index": 2,
           "Severity": 1,
           "Value": 0
         },
         {
           "Direction": "less than",
           "Name": "lower critical",
           "Index": 3,
           "Severity": 1,
           "Value": 0
         },
         {
           "Direction": "less than",
           "Name": "lower critical",
           "Index": 4,
           "Severity": 1,
           "Value": 0
         },
         {
           "Direction": "greater than",
           "Name": "upper critical",
           "Index": 2,
           "Severity": 1,
           "Value": 40
         },
         {
           "Direction": "greater than",
           "Name": "upper critical",
           "Index": 3,
           "Severity": 1,
           "Value": 40
         },
         {
           "Direction": "greater than",
           "Name": "upper critical",
           "Index": 4,
           "Severity": 1,
           "Value": 60
         }
      ],
      "Type": "NCT7802"
    },
```

sanjay-nagesh, 2026-08-25:
Hey @pointbazaar , I'd like to work on this issue if it's still open. I looked into it and it looks like in the `parseThresholdsFromConfig` , when a threshold entry has no `Index` field, it treats it as matching the first sensor (index 1) by default. Then when that sensor sets up its D-Bus interfaces, it tries to add the same "upper critical" interface twice and crashes.

Please let me know if I can work on this. I will first run a test with the same configs you had given and then fix it so a missing Index will only default to sensor 1 when there's no Label.