openbmc/dbus-sensors #28: How to use io-uring

jack206, 2023-10-17:
I want to use io-uring in dbus-sensors, My boost version is 1.78.0, How do I use boost version 1.79.0 alone in dbus-senos？

lcs@ffab0ce87af3:/openbmc/build$ bitbake -s |grep boost
boost                                              :1.78.0-r0
boost-build-native                                  :4.4.1-r0
boost-native                                       :1.78.0-r0
boost-sml                                           :1.1.5-r0
boost-url                                             :git-r0
boost-url-native                                      :git-r0
nativesdk-boost                                    :1.78.0-r0
nativesdk-boost-sml                                 :1.1.5-r0
nativesdk-boost-url                                   :git-r0

lcs@ffab0ce87af3:/openbmc/build$ bitbake -C compile dbus-sensors
WARNING: Layer cmcc should set LAYERSERIES_COMPAT_cmcc in its conf/layer.conf file to list the core layer names it is compatible with.
Loading cache: 100% |###################################################################################################################################################################################| Time: 0:00:00
Loaded 4293 entries from dependency cache.
Parsing recipes: 100% |#################################################################################################################################################################################| Time: 0:00:00
Parsing of 2798 .bb files complete (2797 cached, 1 parsed). 4294 targets, 387 skipped, 0 masked, 0 errors.
WARNING: No bb files in default matched BBFILE_PATTERN_ruijie '^/openbmc/meta-ruijie/'
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "2.0.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "debian-10"
TARGET_SYS           = "arm-openbmc-linux-gnueabi"
MACHINE              = "bmc10b"
DISTRO               = "openbmc-bmc10b"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "arm thumb arm1176jzs"
TARGET_FPU           = "soft"
meta
meta-oe
meta-networking
meta-perl
meta-python
meta-security
meta-phosphor
meta-aspeed
meta-openpower       = "heads/2.12.0:ce7bef12b17859cef0615675e4ad5f6f4f611384"
meta-ruijie
meta-bmc10b
meta-cmcc            = "meta-ruijie:cb21fb129646dc1a6a50e8bf2651f3cfa6c9f48c"

WARNING: The BB_DISKMON_DIRS "ABORT" action has been renamed to "HALT", update configuration
NOTE: Tainting hash to force rebuild of task /openbmc/meta-phosphor/recipes-phosphor/sensors/dbus-sensors_git.bb, do_compile                                                                            | ETA:  0:00:00
WARNING: /openbmc/meta-phosphor/recipes-phosphor/sensors/dbus-sensors_git.bb:do_compile is tainted from a forced run                                                                                    | ETA:  0:00:00
Initialising tasks: 100% |##############################################################################################################################################################################| Time: 0:00:00
Sstate summary: Wanted 5 Local 0 Mirrors 0 Missed 5 Current 408 (0% match, 98% complete)
NOTE: Executing Tasks
ERROR: dbus-sensors-0.1+gitAUTOINC+7dd6443b88-r0 do_configure: meson failed
ERROR: dbus-sensors-0.1+gitAUTOINC+7dd6443b88-r0 do_configure: ExecutionError('/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/dbus-sensors/0.1+gitAUTOINC+7dd6443b88-r0/temp/run.do_configure.175', 1, None, None)
ERROR: Logfile of failure stored in: /openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/dbus-sensors/0.1+gitAUTOINC+7dd6443b88-r0/temp/log.do_configure.175
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['/openbmc/meta-openembedded/meta-oe/recipes-devtools/nlohmann-json/nlohmann-json_3.10.5.bb:do_populate_sysroot', '/openbmc/meta-openembedded/meta-oe/recipes-support/libgpiod/libgpiod_1.6.3.bb:do_populate_sysroot', '/openbmc/meta-phosphor/recipes-extended/sdbusplus/sdbusplus_git.bb:do_populate_sysroot', '/openbmc/meta-phosphor/recipes-phosphor/logging/phosphor-logging_git.bb:do_populate_sysroot', '/openbmc/meta/recipes-core/glibc/glibc_2.35.bb:do_populate_sysroot', '/openbmc/meta/recipes-core/systemd/systemd-systemctl-native.bb:do_populate_sysroot', '/openbmc/meta/recipes-devtools/gcc/gcc-cross_11.3.bb:do_populate_sysroot', '/openbmc/meta/recipes-devtools/gcc/gcc-runtime_11.3.bb:do_populate_sysroot', '/openbmc/meta/recipes-devtools/i2c-tools/i2c-tools_4.3.bb:do_populate_sysroot', '/openbmc/meta/recipes-devtools/qemu/qemu-native_6.2.0.bb:do_populate_sysroot', '/openbmc/meta/recipes-devtools/quilt/quilt-native_0.67.bb:do_populate_sysroot', '/openbmc/meta/recipes-support/boost/boost_1.78.0.bb:do_populate_sysroot', 'virtual:native:/openbmc/meta/recipes-devtools/meson/meson_0.61.3.bb:do_populate_sysroot', 'virtual:native:/openbmc/meta/recipes-devtools/ninja/ninja_1.10.2.bb:do_populate_sysroot', 'virtual:native:/openbmc/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', 'virtual:native:/openbmc/meta/recipes-devtools/pkgconfig/pkgconfig_git.bb:do_populate_sysroot', 'virtual:native:/openbmc/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', 'virtual:native:/openbmc/meta/recipes-devtools/python/python3_3.10.4.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['nlohmann-json', 'libgpiod', 'sdbusplus', 'phosphor-logging', 'glibc', 'systemd-systemctl-native', 'gcc-cross-arm', 'gcc-runtime', 'i2c-tools', 'qemu-native', 'quilt-native', 'boost', 'meson-native', 'ninja-native', 'patch-native', 'pkgconfig-native', 'pseudo-native', 'python3-native', 'linux-libc-headers', 'flex-native', 'texinfo-dummy-native', 'binutils-cross-arm', 'zlib-native', 'zstd-native', 'gnu-config-native', 'libmpc-native', 'gmp-native', 'libtool-native', 'mpfr-native', 'xz-native', 'icu', 'bzip2', 'zlib', 'python3', 'opkg-utils', 'libgcc', 're2c-native', 'shadow-native', 'base-passwd', 'shadow-sysroot', 'base-files', 'systemd', 'sdeventplus', 'phosphor-dbus-interfaces', 'shadow', 'packagegroup-obmc-yaml-providers', 'libcereal', 'libnsl2-native', 'util-linux-libuuid-native', 'libtirpc-native', 'readline-native', 'bzip2-native', 'libffi-native', 'openssl-native', 'sqlite3-native', 'gdbm-native', 'ncurses-native', 'glib-2.0-native', 'attr-native', 'python3-setuptools-native', 'python3-installer-native', 'python3-wheel-native', 'gettext-minimal-native', 'm4-native', 'autoconf-archive', 'util-linux-libuuid', 'readline', 'libxcrypt', 'libtirpc', 'gdbm', 'libffi', 'xz', 'ncurses', 'libnsl2', 'sqlite3', 'openssl', 'util-linux', 'bash-completion', 'zstd', 'libcap', 'libseccomp', 'kmod', 'libpam', 'stdplus', 'function2', 'openpower-ipmi-oem-yaml-provider', 'openpower-debug-collector-yaml-provider', 'phosphor-led-manager-yaml-provider', 'openpower-occ-control-yaml-provider', 'phosphor-software-manager-yaml-provider', 'perl-native', 'libpcre-native', 'gettext-native', 'util-linux-native', 'unzip-native', 'python3-flit-core-native', 'libcap-ng', 'cracklib', 'flex', 'liburing', 'fmt', 'make-native', 'libcap-ng-native', 'libpcre2-native']
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_configure
| NOTE: Executing meson -Db_lto=true -Dadc=disabled -Dcpu=disabled -Dexit-air=disabled -Dexternal=disabled -Dfan=disabled -Dhwmon-temp=enabled -Dintrusion=disabled -Dipmb=disabled -Dmcu=disabled -Dnvme=disabled -Dpsu=enabled -Dtests=disabled...
| The Meson build system
| Version: 0.61.3
| Source dir: /openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/dbus-sensors/0.1+gitAUTOINC+7dd6443b88-r0/git
| Build dir: /openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/dbus-sensors/0.1+gitAUTOINC+7dd6443b88-r0/build
| Build type: cross build
| Project name: dbus-sensors
| Project version: 0.1
| C++ compiler for the host machine: arm-openbmc-linux-gnueabi-g++ -marm -mcpu=arm1176jz-s -fstack-protector-strong -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/dbus-sensors/0.1+gitAUTOINC+7dd6443b88-r0/recipe-sysroot (gcc 11.3.0 "arm-openbmc-linux-gnueabi-g++ (GCC) 11.3.0")
[... 20 lines cut ...]
| Subproject  boost is buildable: NO (disabling)
| WARNING: include_directories sandbox violation!
| The project is trying to access the directory subprojects/boost_1_79_0/ which belongs to a different
[... cut ...]
| Instead of poking directly at the internals the subproject should be executed and
| it should set a variable that the caller can then use. Something like:
|
| # In subproject
| some_dep = declare_dependency(include_directories: include_directories('include'))
|
| # In subproject wrap file
| [provide]
| some = some_dep
|
| # In parent project
| some_dep = dependency('some')
| executable(..., dependencies: [some_dep])
|
| This warning will become a hard error in a future Meson release.
|
|
| ../git/meson.build:65:2: ERROR: Include dir subprojects/boost_1_79_0/ does not exist.

jack206, 2023-10-17:
submit  error

williamspatrick, 2023-10-17:
You can’t use an old release and then expect to be able to update random packages. There are tight dependencies within the image on a particular version of boost. You can’t [easily] update that independently (or at least it isn’t a use case we are going to support).

closed 2023-10-17 by williamspatrick

jack206, 2023-10-18:
I have updated to boost1.79.0, but psusensor is also blocking, it seems that io_uring is not effective, how can I confirm that io_uring is effective?

edtanous, 2023-10-18:
Use a real openbmc release.  You should not be mixing and matching boost.  Use what’s given.

jack206, 2023-10-23:
Has anyone cached sensor values with unordered_map？ Like https://github.com/openbmc/docs/blob/6b367a2f9b5fe3e103a51d584111a7f569073f90/designs/phosphor-hwmon-io-uring.md?plain=1#L4