openbmc/dbus-sensors #29: ADC sensor value is "na" and TempSensor value not update

noman-kwai, 2023-10-19:
I added ADC and TMP75 sensors to my BaseBoard.json file just like this:
`"Exposes": [
        {
            "Index": 0,
            "Name": "AUX3V3",
            "PowerState": "On",
            "ScaleFactor": 0.28147,
            "Thresholds": [
                {
                    "Direction": "greater than",
                    "Name": "upper critical",
                    "Severity": 1,
                    "Value": 3.96
                },
                {
                    "Direction": "greater than",
                    "Name": "upper non critical",
                    "Severity": 0,
                    "Value": 3.63
                },
                {
                    "Direction": "less than",
                    "Name": "lower non critical",
                    "Severity": 0,
                    "Value": 2.97
                },
                {
                    "Direction": "less than",
                    "Name": "lower critical",
                    "Severity": 1,
                    "Value": 2.64
                }
            ],
            "Type": "ADC"
        },
        {
            "Index": 1,
            "Name": "P1V0",
            "PowerState": "On",
            "ScaleFactor": 0.8528,
            "Thresholds": [
                {
                    "Direction": "greater than",
                    "Name": "upper critical",
                    "Severity": 1,
                    "Value": 1.2
                },
                {
                    "Direction": "greater than",
                    "Name": "upper non critical",
                    "Severity": 0,
                    "Value": 1.1
                },
                {
                    "Direction": "less than",
                    "Name": "lower non critical",
                    "Severity": 0,
                    "Value": 0.9
                },
                {
                    "Direction": "less than",
                    "Name": "lower critical",
                    "Severity": 1,
                    "Value": 0.8
                }
            ],
            "Type": "ADC"
        },
        {
            "Index": 2,
            "Name": "P12V",
            "PowerState": "On",
            "ScaleFactor": 0.03701,
            "Thresholds": [
                {
                    "Direction": "greater than",
                    "Name": "upper critical",
                    "Severity": 1,
                    "Value": 14.4
                },
                {
                    "Direction": "greater than",
                    "Name": "upper non critical",
                    "Severity": 0,
                    "Value": 13.2
                },
                {
                    "Direction": "less than",
                    "Name": "lower non critical",
                    "Severity": 0,
                    "Value": 10.8
                },
                {
                    "Direction": "less than",
                    "Name": "lower critical",
                    "Severity": 1,
                    "Value": 9.6
                }
            ],
            "Type": "ADC"
        },
        {
            "Address": "0x48",
            "Bus": "4",
            "Name": "temperature1",
            "Thresholds": [
                {
                    "Direction": "greater than",
                    "Name": "upper critical",
                    "Severity": 1,
                    "Value": 80
                },
                {
                    "Direction": "greater than",
                    "Name": "upper non critical",
                    "Severity": 0,
                    "Value": 70
                },
                {
                    "Direction": "less than",
                    "Name": "lower non critical",
                    "Severity": 0,
                    "Value": 5
                },
                {
                    "Direction": "less than",
                    "Name": "lower critical",
                    "Severity": 1,
                    "Value": 2.5
                }
            ],
            "Type": "TMP75"
        },
        {
            "Address": "0x49",
            "Bus": "4",
            "Name": "temperature2",
            "Thresholds": [
                {
                    "Direction": "greater than",
                    "Name": "upper critical",
                    "Severity": 1,
                    "Value": 80
                },
                {
                    "Direction": "greater than",
                    "Name": "upper non critical",
                    "Severity": 0,
                    "Value": 70
                },
                {
                    "Direction": "less than",
                    "Name": "lower non critical",
                    "Severity": 0,
                    "Value": 5
                },
                {
                    "Direction": "less than",
                    "Name": "lower critical",
                    "Severity": 1,
                    "Value": 2.5
                }
            ],
            "Type": "TMP75"
        }`

In BMC, they can be read by cat CMD 
ADC::~# cat /sys/class/hwmon/hwmon8/in1_input 
         939
TMP75:~# cat /sys/class/hwmon/hwmon6/temp1_input 
             30750

But use ipmitool to send "sensor list" CMD, it return:
`
temperature1     | 31.000     | degrees C  | ok    | na        | 3.000     | 5.000     | 70.000    | 80.000    | na
temperature2     | 30.000     | degrees C  | ok    | na        | 3.000     | 5.000     | 70.000    | 80.000    | na
AUX3V3           | na         | Volts      | na    | na        | 2.635     | 2.962     | 3.639     | 3.966     | na
P1V0             | na         | Volts      | na    | na        | 0.797     | 0.896     | 1.104     | 1.204     | na
P12V             | na         | Volts      | na    | na        | 9.550     | 10.887    | 13.179    | 14.325    | na
` 
ADC sensor value is "na" 
TempSensor value is ok but not update.
 
I found some errors by  "journalctl | grep hwmon" CMD:

` kernel: lm75 4-0048: hwmon6: sensor 'tmp75'
 kernel: lm75 4-0049: hwmon7: sensor 'tmp75'
 hwmontempsensor[266]: error getting  SpecialMode status No route to host
 hwmontempsensor[266]: Error calling mapper
 hwmontempsensor[266]: failed to find match for 4-0048
 hwmontempsensor[266]: failed to find match for 4-0049
 adcsensor[264]: failed to find match for /sys/class/hwmon/hwmon8/in3_input
 adcsensor[264]: failed to find match for /sys/class/hwmon/hwmon8/in1_input
 adcsensor[264]: failed to find match for /sys/class/hwmon/hwmon8/in2_input`

Is there some  configuration missing  or operation error ？

I‘m a beginner of openbmc, Please give me some advice.

Thank you!

Kostr, 2023-10-19:
Do ADC sensors have value or na on the webui sensors page?

I've also noticed, that you've added `"PowerState": "On"` to your ADC  sensors. The affect of this is that the sensors are not checked if the main host is not powered on. Was the host on or off when you've executed the `ipmitool sensor` command?

And it is very hard to see anything from your output, I suggest you to properly format it.

noman-kwai, 2023-10-19:
Thanks for your reply.
> Do ADC sensors have value or na on the webui sensors page?
>  ADC sensors value is also na on the webui.
> I've also noticed, that you've added `"PowerState": "On"` to your ADC  sensors. The affect of this is that the sensors are not checked if the main host is not powered on. Was the host on or off when you've executed the `ipmitool sensor` command?
> The host was off.
 Now I comment out  the "PowerState": "On" line, ADC sensors work fine.
 But TempSensors still can't update the value,it remains the value when BMC PowerOn.Is it also because the configuration statement of the json file is incorrect.Where I can find more Docs
> And it is very hard to see anything from your output, I suggest you to properly format it.

Thank you for your advice. I am not familiar with the usage yet, and I will pay attention to the format later.

noman-kwai, 2023-10-19:
Thanks for your reply.
> Do ADC sensors have value or na on the webui sensors page?
> 
ABC senses value is also na on the webui.
> I've also noticed, that you've added `"PowerState": "On"` to your ADC  sensors. The affect of this is that the sensors are not checked if the main host is not powered on. Was the host on or off when you've executed the `ipmitool sensor` command?
> 
The host was off.
Now I comment out  the "PowerState": "On" line, ADC sensors work fine.
 But TempSensors still can't update the value,it remains the value when BMC PowerOn.Is it also because the configuration statement of the json file is incorrect.Where I can find more Docs

> And it is very hard to see anything from your output, I suggest you to properly format it.

Thank you for your advice. I am not familiar with the usage yet, and I will pay attention to the format later.

Kostr, 2023-10-19:
The data from the `/sys/class/hwmon/hwmonX/tempY_input` file should be the same as the sensor value in the ipmitool/webui output.

- If they are not the same, you are probably looking at the wrong `/sys/class/hwmon/hwmonX/tempY_input` file
- If they are the same, and the value does not change, then I guess the real tempreature doesn't change. What makes you think that the mesaurement should change its value?

Kostr, 2023-10-19:
Are you sure that this fan cools this particular TMP75 ?

You've said
> But TempSensors still can't update the value,it remains the value when BMC PowerOn

You can check this theory.
Unplug your fan, wait some time time to be sure, that the value doesn't change. If that is the case, without plugging the fan back, you can reboot bmc, and check that on the next boot the sensor value is way higher. Is that is case?

noman-kwai, 2023-10-20:
> Are you sure that this fan cools this particular TMP75 ?
> 
Yse, I'm sure. 

> You can check this theory.
> Unplug your fan, wait some time time to be sure, that the value doesn't change. If that is the case, without plugging the fan back, you can reboot bmc, and check that on the next boot the sensor value is way higher. Is that is case?
> 

I tried this theory, the next boot sensor value  not change anymore.
Is it because the sensor drive is not working properly?